Rohde-Schwarz/TrustedGRUB2 #2: Which PCR should I use for sealing NVRAM and how to compute hashes manually

momiji, 2015-03-25:
Hello again,

I've made some research and I am in the move to integrate trustedgrub2 with a modified vesion of tpm-luks (improvements + rhel7 support), you may find in my github repos.

My next questions are about how one shuold configure trustedgrub2 with tpm-luks.
1. which PCR should we use to seal the NVRAM
2. how to compute the expected PCR values

For 1., it seems I'll need to seal with all PCR values, 8 to 12, but should I add other numbers (like 0 etc.) ?  May is it possible to compute 
I will also probably add a password to the sealed NVRAM, as TPM has a built-in security that locks the TPM to prevent brute force attacks.

For 2, in the original tpm-luks, there is a shell script which computes the permissions file to seal the NVRAM. This is basically a simple text file containing a list of "r <PCR INDEX> <HASH>". Unfortunately this script was made for grub1, and I'll need to update it to deal with grub2 config files. May be you can help in identifying the elements I should compute the hash for, and I hope this is possible to do.

I'll be happy to share my experience on all this with you as it seems nobody has ever tried to do all this before :)

neusdan, 2015-03-25:
Hi,

"which PCR should we use to seal the NVRAM"

Thats in your decision. You should take a look into the TPM specification to see whats actually measured in the different PCRs. And then you must decide to which of them you want to seal your data. 

You will find the specification here (especially page 41 is important):
https://www.trustedcomputinggroup.org/resources/pc_client_work_group_specific_implementation_specification_for_conventional_bios

Example:
The BIOS  is measured into PCR 0. So if you have sealed your data to PCR 0 and you update your BIOS, the value of PCR 0 will change and you are no longer able to unseal your data. This are things you have to adjust to your needs.

For 2:
I'm not familiar with tpm-luks. I would like to help with this, but i need more information what you are actualy doing. So at least i need to know which PCR values you are trying to precompute.

momiji, 2015-03-27:
Hello,

I'll try to explain better - English is not my mother tong, I'll do my best.
And thanks a lot for your help, and taking the time to read an try to understand my questions.

You'll find all my work here: https://github.com/momiji/tpm-luks, where I tried to explain as much as possible everything in the README.md documentation.

### tpm-luks
tpm-luks is a module for dracut (loaded in initramfs), that allows to open LUKS partitions using the TPM NVRAM to get the LUKS key.

In the original documentation, https://github.com/shpedoikal/tpm-luks, it already refers to TrustedGRUB to seal the NVRAM with some PCR, that are computed by TrustedGRUB. What I understand is that PCR 0-7 are computed by the BIOS itself and 8-15 by TrustedGRUB2. Can you confirm ?

### PCR in original tpm-luks / TrustedGRUB combination

The file used to configure which PCR to use to seal NVRAM https://github.com/shpedoikal/tpm-luks uses PCR 4 5 8 9 12 14,

>
```
# PCR 4  - record measurement of trusted grub in the MBR as measured by
#          the BIOS
# PCR 5  - record the IPL config - this is a record of the boot device
#          chosen by the user
# PCR 8  - The first 512 bytes of grub stage2
# PCR 9  - The rest of stage2
# PCR 12 - record trusted grub's config file command lines - note that
#          this includes the kernel command line, so any change to the
#          kernel command line will prevent releasing the NVRAM secret!
# PCR 14 - record the files that trusted grub launches - kernel and
#          initramfs
PCRS="4 5 8 9 12 14"
```

### PCR in new tpm-luks/TrustedGRUB2 combination

It seems that the computed PCR are not the same in the new TrustedGRUB2:

>
```
# PCR 8 First sector of TrustedGRUB2 kernel (diskboot.img)
# PCR 9 TrustedGRUB2 kernel (core.img)
# PCR 10 Everything that is loaded from disk (grub2-modules, Linux-kernel, initrd, ntldr, etc.) // TODO: fonts, themes, locales
# PCR 11 contains all commandline arguments from scripts (e.g. grub.cfg) and those entered in the shell
# PCR 12 LUKS-header
```

My impression is that I need to use 4, 5 + 8 to 12 to mimic original behavior, which answers my first question.

### Compute expected PCR values

In the original tpm-luks, there is the script https://github.com/momiji/tpm-luks/blob/master/tpm-luks/tpm-luks-gen-tgrub-pcr-values that computes the expected PCR hashes, directly by reading the menu.lst file, stage2 file...

What I want to do is the same, creating a new tpm-luks-gen-tgrub2-pcr-values that does the same, but for new TrustedGRUB2, with grub.cfg instead of old menu.lst, ...

This is where my second question comes in, as I would like to automatically compute the PCR the same way as TrustedGRUB2 does.

This way I would be able to seal the NVRAM, reboot, then TrustedGRUB2 updates the NVRAM and gives the same values, allowing tpm-luks to read them and open the LUKS partitions.

neusdan, 2015-03-28:
Hi,

> What I understand is that PCR 0-7 are computed by the BIOS itself and 8-15 by TrustedGRUB2. Can you confirm ?

yes exactly.

### Precomputation:

Here are my ideas how to do the precomputation:

> PCR 8 First sector of TrustedGRUB2 kernel (diskboot.img)

This is quite easy.
```
dd if=/dev/sdX bs=512 skip=1 count=1 | sha1sum
```
Replace sdX with the device where TrustedGRUB2 is installed. The resulting PCR8 value should than be:

```
PCR8 = sha1sum( 0000000000000000000000000000000000000000 || SHA1 from command above )
```

> PCR 9 TrustedGRUB2 kernel (core.img)

This is more dificullt. First you should determine the size of your core.img. This size is contained in the diskboot.img. 

```
dd if=/dev/sdX bs=512 skip=1 count=1 | hexdump -C
````

Analyze the hexdump. Some of the last bytes contain the number of sectors of the core.img. At the moment if have no environment here to test, but i think you should see it. If you have the number you can measure the core.img:

```
dd if=/dev/sdX bs=512 skip=2 count=NumberOfSectorsCoreIMG | sha1sum
```

Replace sdX and NumberOfSectorsCoreIMG accordingly.

> PCR 10 Everything that is loaded from disk (grub2-modules, Linux-kernel, initrd, ntldr, etc.) // TODO: fonts, themes, locales

grub2-module measurement would be quite difficult. For now i have no idea how to do this properly. So the easiest solution is to place all modules that are needed in the core.img so that they are measured already in PCR9. To do this you can specify modules that should be placed in the core.img to the "grub-install" command. For example:

```
./grub-install --modules="part_msdos biosdisk ntfs minicmd keylayouts gettext tpm" ...
```

Measurement of kernel and initrd is easy again. Just extend the SHA1 hash of those to files.

> PCR 11 contains all commandline arguments from scripts (e.g. grub.cfg) and those entered in the shell

If you have a simple grub.cfg this is also easy. Just hash the strings that are executed in the order that they are executed.

> PCR 12 LUKS-header

You don't need this if you are not using the "cryptomount" command.

momiji, 2015-03-30:
Thanks a lot, I'll check this ASAP. I have a dev platform with TPM installed, and I'll come back to you with the results.

closed 2015-07-22 by neusdan