Rohde-Schwarz/TrustedGRUB2 #3: spec file for building RPM

momiji, 2015-03-25:
It's me again...

You might be interested in the xtra/TructedGRUB2.spec file I've made to build the rpm.

OK, this is probably a very bad spec file, I learned about building RPM only 3 days ago, and my only objective was to have a very simple RPM to deploy on servers without having to manually build localy.

I found the spec file of grub2 2.0.2 on the web, but it seems much to complex for my poor little experience.

Also, I didn't understand the difference between grub and efi (on my hardware grub.cfg files are in /boot/efi/... instead of only /boot, I still haven't understood why. Should we build something special to have EFI support ? or should I just forget about it, and if so, what should I do with the 2 partitions /boot and /boot/efi ?

My comprehension was that EFI needed to be signed, so maybe TrustedGRUB2 cannot work in EFI mode.

Thank's a lot!

neusdan, 2015-03-25:
Hi,

I've only implemented the TPM features for GRUB2 in MBR/Legacy Mode. It's also possible to do all the TPM stuff like measurements, unsealing etc. in EFI-Mode, but thats a complete different API and would require a lot of work. From my side, it's not planned for the moment, but maybe someone contributes it in the future.

Obviously you are running TrustedGRUB2 in MBR-Mode, otherwise the TPM features should not work. So i don't know why your grub.cfg is in /boot/efi, it actually make not much sense, seems to be some misconfiguration.

I'm not using Red Hat and not familiar with those .spec files, but you can send a pull request and i will add it.

momiji, 2015-03-27:
Hello, well I finally understood where EFI comes from.

When I installed RHEL7 on my VirtualBox (I know there is no TPM but TrustedGRUB2 does work, of course), it installs without EFI, as secure boot is disabled by default in the BIOS. But it is possible to change it, and I managed to re-install with EFI, just to play with it.

When I installed RHEL7 on my tset computer, it installs automatically with EFI, and chaging the secure boot option in the BIOS didn't change anything. I finally found that I need to change the boot order to disable the EFI boot devices. Then, RHEL7 installs without EFI, and everything worked like a charm (well not completely, i'll write another post).

When playing with both installs, I finally understood that in EFI mode, there are 2 boot partitions (sda1 and sda2 containing /boot/efi and /boot). On the contrary, in non-EFI mode, there is only 1 boot partition containing the /boot. My guess is that the RHEL CDROM contains a system that allows both boot, one in EFI mode, one without. As proof, the grub gui is not the same.

I'm not fluent in linux enough to rebuild the first boot partition as a non-EFI one, so my best option was to reinstall without EFI support at all.

Thanks a lot for your help! - this can be closed

closed 2015-03-28 by neusdan