Rohde-Schwarz/TrustedGRUB2 #4: UEFI booting mechanism supporting

baaltommysu, 2015-04-20:
Hi, I am testing to make TrustedGRUB2 working with a UEFI booting machine. I have tested TG2 with an old machine of legacy BIOS booting option, it works correctly without any problem. 

Then I turned off the 'legacy BIOS supporting' from the machine's setting. This time it cannot install into the device again. Here is the output error.

    /root/TrustedGRUB2/sbin/grub-bios-setup: warning: this GPT partition label contains no BIOS Boot Partition; embedding won't be possible.
    /root/TrustedGRUB2/sbin/grub-bios-setup: warning: Embedding is not possible.  GRUB can only be installed in this setup by using blocklists.  However, blocklists are UNRELIABLE and their use is discouraged..
    /root/TrustedGRUB2/sbin/grub-bios-setup: error: will not proceed with blocklists.

My question is, does TG2 support UEFI boot mechanism? Or it only support the legacy BIOS booting option? If it support UEFI, how can I install it? 

Thank you very much.

neusdan, 2015-04-20:
Hi, currently TrustedGRUB2 only supports TPM features in legacy BIOS/MBR boot mode. But UEFI booting should still work, as GRUB2 supports UEFI.

I'm wondering why you are calling grub-bios-setup. I think this is the legacy mbr mode install.

baaltommysu, 2015-04-21:
Hi, I only run the install command as instructed in the README.md, this is the command I run: 

   ./INSTALLDIR/sbin/grub-install --directory=INSTALLDIR/lib/grub/i386-pc /dev/sda 

And the output is the one I showed above. 

Is there some other argument I should add in this command? Or there is some other command I should execute?

Another question, if TrustedGRUB2 only support TPM feasures in legacy BIOS/MBR boot mode, what will happen if TrustedGRUB2 is installed and only UEFI boot mode is allowed? The TPM will not be used? No module will be measured and extended into PCRs?

Thank you very much

neusdan, 2015-04-21:
You probably have to add `--target=x86_64-efi` to the grub-install command. But i have never installed GRUB2 in UEFI-Mode. So you should look for GRUB2 documentation on that topic.

In theory, if you install TrustedGRUB2 in UEFI-Mode, the TPM features are just disabled and  TrustedGRUB2 acts as a normal GRUB2. But i've never tested it.

baaltommysu, 2015-04-23:
I tried to add --target=x86_64-efi to the grub-install command, but the result is the same, I re-paste the output here: 

    bash /root/TrustedGRUB2/sbin/grub-install --target=x86_64-efi --directory=/root/TrustedGRUB2/lib/grub/i386-pc /dev/sda
    /root/TrustedGRUB2/sbin/grub-bios-setup: warning: this GPT partition label contains no BIOS Boot Partition; embedding won't be possible.
    /root/TrustedGRUB2/sbin/grub-bios-setup: warning: Embedding is not possible.  GRUB can only be installed in this setup by using blocklists.  However, blocklists are UNRELIABLE and their use is discouraged..
    /root/TrustedGRUB2/sbin/grub-bios-setup: error: will not proceed with blocklists.

Any idea why and how to fix it?

neusdan, 2015-04-23:
You have compiled TrustedGRUB2 for i386-pc platform. 

Try to rebuild with the following configure options: `--with-platform="efi" --target="x86_64"` or `--target="i386"` depending on your platform. 

You have also to adjust the `--directory` parameter in the `grub-install` command.

Maybe the GRUB2 mailing list is the better place to ask. At the moment i can only support legacy bios installations and i have no experience with GRUB2 in UEFI-mode.

baaltommysu, 2015-04-23:
Hi, thank you for your answer. 
I tried to compile TrustedGRUB2 with --with-platform="efi" --target="x86_64", and then make will always fail with the following error: 

    grub-core/disk/luks.c:33:30: fatal error: grub/machine/tpm.h: No such file or directory
    #include <grub/machine/tpm.h>
                              ^
    compilation terminated.

This error is caused by the argument of --with-platform=efi. 
I will also try to look for answers from other where, thank you very much for your time.

neusdan, 2015-04-23:
Hi,

i understand where this error comes from and could fix that. But because TrustedGRUB2 does not support UEFI at the moment, it doesn't make much sense.

So, if you wan't to use TrustedGRUB2 use legacy mode. If you wan't to have a UEFI Bootloader without TPM features use GRUB2.

baaltommysu, 2015-04-29:
OK, thank you very much. I will close this issue.

closed 2015-04-29 by baaltommysu