Rohde-Schwarz/TrustedGRUB2 #5: Failure to cryptomount with key (sealed or not)

albal, 2015-05-21:
Hi,

I am implementing an embedded system on x86 and was looking to TrustedGRUB2 to perform unlock of a luksFormat partition.  This partition is actually on a RAID array.  I think I have all the correct modules loaded in grub.  `ls` shows the devices and if I use a pass-phrase I can open the encrypted partition.  

However, when I use the `-k` option it fails, either using unsealing or not (I change the key accordingly to one that is sealed or clear).  I put some line numbers into `grub-core/disk/luks.c` in the `grub_fatal("luks_recover_key_failed")` lines to be able to identify where the function was failing but I am unclear to actually what is happening.  My thoughts are the file is not being read.  It would appear to drop through to the last fatal on line 601.

I tried on a disk that had both `ms_dos` and `gpt` layouts.  For reference I build TrustedGRUB2 with the following options which is a derivation of what the buildroot team use for GURB2:

```text
TRUSTEDGRUB2_TARGET = i386
TRUSTEDGRUB2_PLATFORM = pc

TRUSTEDGRUB2_CONF_OPTS = \
        --prefix=$(HOST_DIR) \
        --disable-nls \
        --target=$(TRUSTEDGRUB2_TARGET) \
        --with-platform=$(TRUSTEDGRUB2_PLATFORM) \
        --disable-grub-mkfont \
        --enable-efiemu=no \
        --enable-liblzma=no \
        --enable-device-mapper=no \
        --enable-libzfs=no \
        --disable-werror
```

If I try to unseal the key then I am faced with the following error:
```text
Attempting to decrypt master key...
tcg_passThroughToTPM: asm_tcg_passThroughToTPM failed: 10000
Aborted. Press any key to exit.
```

If I try to use the key in the clear I get:

The grub config is I am using is:
```text
set default="0"
set timeout="5"

menuentry "Bank 1" {
        cryptomount md/0,gpt1 -k (hd0,gpt2)/keyfile -s
        linux (hd0,gpt5)/boot/bzImage
}
```

If I change the keyfile name in grub so it is mismatched with the file on the disk I get the correct error that the file is not found, ie:
```text
error: file `/keyfil' not found
```

But somewhere along the line the keyfile is not being pass through.  Do you have any pointers for me as to where I am going wrong?

Thanks,
Al

neusdan, 2015-05-22:
Hi,

can you set `debug=2` before issuing the cryptomount command and post the output here?

> If I try to use the key in the clear I get:

i think you forgot to post the output ;)

albal, 2015-05-22:
Hi,

I updated my original post to include the missing output from clear key use.  I'll post back once I have recompiled with debug support.  Buildroot turns off debug by default :-)

Cheers,
Al

albal, 2015-05-22:
Hi,

I have built TrustedGRUB2 with debug.  Though I found I had to enter `set debug=all` to get any output - the last screen full is shown below:

[image]

For search purposes I write:
```text
disk/luks.c:434: Trying keyslot 0 
disk/luks.c:457: PBKDF2 done
disk/luks.c:526: candidate key recovered
disk/luks.c:552: bad digest
```

Above you can see I added the line number into the many `luks_recover_key failed.` messages to see where it was exiting exactly.  I will investigate PBKDF2 a little more while I wait for your feedback.

Kind regards,
Al

neusdan, 2015-05-22:
It's failing at `disk/luks.c:552: bad digest`. So for me it looks like that your keyfile does not fit the LUKS keyslot on the partition.

albal, 2015-05-22:
I used a script to do all the tpm and crypt operations.  I use the same reference for the keyfile and generate it thus:
```text
dd bs=512 count=8 if=/dev/urandom of=${keyfile}
```
I will try using a static file and see where I am from there.  Thanks for your reply.

neusdan, 2015-05-22:
8*512 byte keyfile seems to be big ;)

neusdan, 2015-05-22:
Should be 32 byte if you are using a 256 bit cipher

albal, 2015-05-22:
Makes perfect sense now you mention it. :-/ Very grateful for your help. 

Also I intend to submit TrustedGRUB2 as a package for Buildroot if you are okay with that?  Unless you want to do it yourself.  I'm happy to share with you the config files.

neusdan, 2015-05-22:
You're welcome! I'm fine with the submitting

Mike-Joseph, 2015-05-26:
I believe that the length of the keyfile (or sealed payload, as the case may be) should not need to match the native LUKS keyslot size due to the application of PBKDF2 hashing.  In fact, I think that the problem described herein is an example of the out-of-bounds memory access error that I have reported in issue #6.

albal, 2015-05-28:
Hello Again :-)

I am still having an issue.  This time it with unsealing the key via cryptomount.  I am using tpm-tools: tpm_sealdata to seal the keyfile.  I verify the data can be unsealed using tpm_unsealdata - that works as expected.  However when TrustedGRUB2 goes to passthrough the function to the TPM via asm_tcg_passThroughToTPM it fails:

```text
tcg_passThroughToTPM: asm_tcg_passThroughToTPM failed: 10000
```

Looking at the TPM 1.2 Spec the error code is shifted 16 bits and indicates general failure.  I notice in [tpm.S](https://github.com/Sirrix-AG/TrustedGRUB2/blob/88b2f1c9c1a29e34ae6b827ff2e716acf647aa78/grub-core/kern/i386/pc/tpm/tpm.S#L210) there are still some todo items.  Is it possible these operations are not compatible with my TPM.  I am using an [Infineon 9655V](http://www.supermicro.com/manuals/other/datasheet_AOM-TPM-9655V.pdf) version 1.2.4.32.

It is possible I can get someone to look further into the code (my only experience of assembler is on micro controllers).  If you have any direction or a solution it would be much appreciated.

Thanks in advance,
Al

neusdan, 2015-05-28:
Hi,

TCG_GENERAL_ERROR seems to be hard to debug. You could add some debug output to track at which point it is failing, i.e. which TPM command fails. To see if it is TPM_Unseal or TPM_GetRandom for example. But i'm not sure if this will really help. 

The BIOS provides the Int1A interface. So it might be a bug in there and could explain why it works with tpm_unsealdata and not over the Int1A interface because here are two different interfaces used.

petrikah, 2015-06-10:
Hi,

I believe that tpm-tools produces sealed key in different format than GRUB expects.
tpm-tools tpm_sealdata produces the output in format:

-----BEGIN TSS-----
-----TSS KEY-----
<UUENCODED KEY>
-----ENC KEY-----
Symmetric Key: AES-256-CBC
<UUENCODED KEY>
-----ENC DAT-----
<UUENCODED KEY>
-----END TSS-----

However, in tpm.c:

	/* TPM_UNSEAL Incoming Operand */
	struct {
		grub_uint16_t tag;
		grub_uint32_t paramSize;
		grub_uint32_t ordinal;
		grub_uint32_t parentHandle;
		grub_uint8_t  sealedData[inputSize];
		grub_uint32_t authHandle;
		grub_uint8_t  nonceOdd[TPM_NONCE_SIZE];
		grub_uint8_t  continueAuthSession;
		grub_uint8_t  parentAuth[TPM_AUTHDATA_SIZE];
		grub_uint32_t dataAuthHandle;
		grub_uint8_t  dataNonceOdd[TPM_NONCE_SIZE];
		grub_uint8_t  continueDataSession;
		grub_uint8_t  dataAuth[TPM_AUTHDATA_SIZE];
	} __attribute__ ((packed)) *unsealInput;

neusdan, 2015-06-10:
Hi,

I had a look into the tpm_sealdata code. You're right. They are using a different format. They create a new RSA key for each tpm_sealdata execution. 

TrustedGRUB2 at the moment only supports unsealing of BLOBS that are sealed directly with the SRK.

But pull requests are welcome ;-)

albal, 2015-06-11:
Hi Dan,

Any chance you can share your tpm sealing tool? I've been in touch with Kent Yoder (of TrouSers / tpm-tools) and I think he might be prepared to add an option into tpm-tools to seal in a format that TrustedGRUB2 would parse.   You can contact me via the email address on my profile page and I also sent a request yesterday to info@sirrix.com - but maybe that wouldn't find you.  I think both solutions would give ultimate flexibility to the end user.

Many thanks in advance,
Al

neusdan, 2015-06-11:
Hi,

Unfortunately there are no plans at the moment to open source our TPM library. So if tpm-tools from TrouSers could be adjusted that would be great. There is no magic behind it and everything that is needed for this is already there in the TrouSers library. The tpm-tools just have to output a sealed blob to a file. That should be quite easy.

petrikah, 2015-06-12:
Hi,

I tried TPM/J, Java based solution, failed with error code 29. Here some test logs below:
sealedData seems to be in TPM_STORED_DATA format with 256 bytes encrypted TPM_SEALED_DATA

tcg_passThroughToTPM
unsealInput parameters:
tag:                 c300
paramSize:           74010000
ordinal:             18000000
parentHandle:        40
sealedData:
01 01 00 00  // ver
00 00 00 00  // sealInfoSize
00 00 01 00 // encDataSize
41 16 26 c0 99 72 37 fe b4 92 28 6b 96 38 f9 09 9b 80 30 eb b0 e6 7d 33 8b e6 21 aa 11 d3 7e 1e 
72 07 12 e9 98 13 af 80 05 9f 77 6a 22 59 46 4c de c6 d3 78 ce f7 69 eb 71 c9 94 cb 7c cd f1 b3 
32 35 66 d4 c9 d8 e7 fb 69 80 b6 ed 11 a0 b3 a3 ac ce 37 12 7f 35 20 43 4d 15 82 df ae 4b 85 d0 
1e 1c 2f 6a ca b6 b7 d3 f6 4a f3 52 be 44 1f 61 85 3d 4c 6b 2b 49 53 ff 41 b1 ef fe 59 bf 80 2d 
46 9f 4f ca 63 46 71 45 01 a8 6f be a4 27 df 45 f4 65 ae 01 3a f4 c2 92 14 c0 b7 91 83 86 1f b3 
f1 65 a4 e4 ec 77 81 1b 67 e2 31 9a 30 7f 01 b3 75 90 ca bd 29 60 5f 0a 91 e0 c6 18 b3 21 d1 c8 
6e c7 4d c3 29 51 c7 8b 96 73 fd e0 a0 c5 86 8d e5 43 de a5 42 44 3f 1f 67 59 79 c6 5f 98 39 41 
e3 c4 c2 fe 85 a7 0b 7b 5d f2 e8 9e 74 27 6f c9 1d 70 e9 ad 10 53 8b b1 37 33 65 8e 0a 8a e4 30
authHandle:          35273404
nonceOdd:
39 02 dd 7a a5 36 4e 2f a5 10 eb eb 09 7d 64 91 ae 10 33 ca
continueAuthSession: 0
parentAuth:
ac a4 4c ff 49 55 39 06 be 77 c7 11 e6 59 22 e1 a5 c6 c5 29
dataAuthHandle:      b46f8e00
dataNonceOdd:
39 5b 8a 8e 5e f1 28 ea 93 c9 aa 92 80 c3 3b 69 75 d7 42 30
continueDataSession: 0
dataAuth:
34 f2 6d ac c8 8d 30 76 e0 9f c8 26 0e 76 8f b2 96 6b 30 94
grub_TPM_unseal: Unsealing failed: 29

neusdan, 2015-06-12:
Hi,

can you describe the arguments you have used? I guess you have created the file with this:

java edu.mit.csail.tpmj.tools.TPMSeal <dataFile> <keyHandle> <keyPwd> <dataPwd> [pcrNums...]

petrikah, 2015-06-12:
Hi,

Yes but without PCRS since TPM/J for some reason returns error code 3

root@aaa-desktop:~/tpm/tpmj/tpmjdemo# java edu.mit.csail.tpmj.tools.TPMSeal pass32 SRK "" courier

Parsing command-line arguments ...

Using SRK to seal.
keyPwd = , Encoded (All zeros) = 0000000000000000000000000000000000000000
dataPwd = courier, Encoded (Infineon/Vista (SHA1 of UTF16LE without null terminator) = 1ce90222dd3af249d68982084085cd7df2a3fd88

*** Reading Data *** ... 

31 32 33 34 35 36 37 38 39 30 31 32 33 34 35 36 37 38 39 30
31 32 33 34 35 36 37 38 39 30 31 32

*** Reading PCR configuation to seal data to *** ... 
PCRs will NOT be used as a condition for unsealing

*** Sealing Data *** ... 
Encrypted data structure =
TPM_STORED_DATA: ver: edu.mit.csail.tpmj.structs.TPM_STRUCT_VER: 01 01 00 00
sealInfo: null
encData (256 bytes): 
78 df d0 c0 38 bd 57 88 a8 24 e7 54 d9 01 04 72 8f 59 e9 3c
b0 37 92 a1 f9 77 0b b7 8d 06 72 fb ef 0c cd ce 29 fe cf 96
88 76 ed 9c 97 a3 f7 f4 a1 67 77 e8 7f bf eb 68 eb 25 9c 18
4e 75 c6 9f 09 1f fb f8 c6 47 fe 9e 7a 17 19 26 86 91 98 c5
45 b1 6f 79 33 1a 07 e0 14 21 a3 2d 8b d1 13 c9 5d c3 5f b2
e9 11 50 c8 12 9e 94 70 f3 c2 7b 73 8e c6 74 08 81 a7 e3 90
a5 21 41 0e 83 87 fd f4 d5 1c 87 29 eb ca 87 40 1c 47 3f c4
b8 20 69 76 3a 69 64 ef 24 cd 6a 82 16 73 aa bf 48 a0 6e 91
46 a2 6b 32 f3 b9 d4 db 3d c6 74 49 44 b4 2d 9b b6 f2 64 a5
d1 e6 32 37 26 da db 80 d5 0d 41 d8 16 74 2b 45 ea a1 eb 22
6b ec 23 ff 77 5b 5c 12 3b 2e 6e c3 0f 10 6f be 8b ca 89 33
7f 33 74 ce 83 e9 3c b2 1c 5e 11 80 a7 58 83 84 a7 af 0f ac
20 02 03 bf 5b 07 70 60 56 ad 07 41 97 df 71 ee

Writing to file pass32.sealed
DONE.

*** Testing Unseal *** ...

31 32 33 34 35 36 37 38 39 30 31 32 33 34 35 36 37 38 39 30
31 32 33 34 35 36 37 38 39 30 31 32

root@aaa-desktop:~/tpm/tpmj/tpmjdemo#

neusdan, 2015-06-12:
I think the problem is the data password. This is currently not supported. Please try setting it also to the well known secret.

petrikah, 2015-06-12:
Hi
WOW!! Thanks for your help!!!
java edu.mit.csail.tpmj.tools.TPMSeal pass32 SRK "" ""

Works!

neusdan, 2015-06-12:
Nice! Happy to hear that ;-)

albal, 2015-07-01:
After some testing (without using PCRs) I am happy to announce that the maintainers of tpm-tools have modified tpm_sealdata to output in a raw format that can be read by TrustedGRUB2.  I have successfully used this function to seal the LUKS keys post install of my system.

You can find it at: https://github.com/shpedoikal/tpm-tools/tree/tpm-sealdata-raw

However, I tired sealing with PCR 11 (config files) and was under the impression that `set default="0"` would be ignored from hashing of the grub.cfg file.  Maybe I read it somewhere else with another version of Grub1.  Is is possible to be able to switch boot entries in grub.cfg and still use PCR 11 to has h the rest of the contents?

I am about to try with other PCRs as maybe this functionality is not yet working correctly in tpm-tools tpm_sealdata.

Thanks,
Al

petrikah, 2015-07-01:
Hi,

I have experimented with TPM/J and succeeded in sealing keyfile with any PCRs in Infineon tpm 1.2 chip. (Trick was to implement PCR_INFO_LONG and TPM_STORED_DATA12  java classes)
 
I noticed the problem in using cryptomount command is that PCRs 10 & 11 cannot be used to seal keyfile afterwards because kernel- and initrd commands that executes after cryptomount also affects those registers. So question is how to make sure that any modification to grub, grub.cfg or kernel command line parameters (e.g., break) or running any command cause PCRs change so that unsealing keyfile with cryptomount does not work.

I suppose that PCR 10/11 are not modified after kernel starts executing from initramfs so maybe initramfs would be a good place for scripts that open luks container. Security must then be enhanced further with grub> measure initrd and vmlinuz into some PCR and use that in addition to seal the keyfile. 

tpm-tools is great news as it may be more elegant to put into initramfs than a bundle containing TPM/J jar and JRE :-)

albal, 2015-07-02:
For me without being able to use PCR 11 on just the grub.cfg file it is not possible to change select of boot entries.  I did try adding `"set"` to the list of ignored args along with `"menuentry"` in [execute.c](https://github.com/Sirrix-AG/TrustedGRUB2/blob/5dc50ffafcfac943330d695283d46379f5f49991/grub-core/script/execute.c#L953) and for my setup without PCR11 in place it is trivial to to access the encrypted partition.

In my application the user can load different operating system images into 3 partitions and within the application they can select which one they can boot.  This is simply done by changing `set default`.

neusdan, 2015-07-06:
Hi,

what if we change PCR 10 to contain only the grub2-modules measurement and introduce a new PCR 13 for linux kernel, initrd, etc.

Then you can use PCR 10 and do the measurement of grub.cfg, kernel and initrd separately in grub.cfg, i.e. 
`measure /boot/grub/grub.cfg 14`
`measure /boot/vmlinuz 14`
`measure /boot/initrd 14`
`cryptomount ....`

@albal 
If you change boot entries and therefore different OS images are loaded MUST result in different PCR values in my in opinion. To support that you should use 3 different keyfiles.

albal, 2015-07-08:
I am closing this issue as I can unseal keys sealed with another opensource tool (tpm-tools).  I have rethought my design, currently untested, which is to chain load a second TrustedGRUB2.  The first TrustedGRUB2 would only have one boot option and sealed key against the config and commands.  It would then unlock the partition where the slave grub is installed and where the keys for other encrypted drives are stored.  After a few hours thought I don't see any issues with this and allows me what I need to do with different boot selects at the users bequest.

@Mike-Joseph @petrikah and @neusdan thanks for all your help, guidance and contribution.

All the best,
Al

closed 2015-07-08 by albal