Rohde-Schwarz/TrustedGRUB2 #9: Kernel measurement does not measure the buffer that's used

mjg59, 2015-07-20:
grub-core/loader/i386/linux.c appears to read the kernel from disk into a series of buffers for later execution, and then reads the kernel again to perform a measurement. A sufficiently malicious storage device might provide a backdoored kernel on the first read attempt, followed by the correct kernel on the second read attempt. The measurement would then appear correct.

alexanderkjeldaas, 2015-08-02:
@mjg59 Have you looked at other parts that read code from disk to see if this bug exists in other places?

mjg59, 2015-08-03:
I'm afraid I haven't - I noticed this one because I made exactly the same mistake in the UEFI Secure Boot implementation.

neusdan, 2015-08-04:
This bug also exists in other places. For example the GRUB2 modules measurement or ntldr / chainloader measurement.

I'm at the moment on vacation. I'll try to fix this afterwards.

PR #33 (2016-01-02): Measure buffer that is used
Before this fix everything that was measured from disk was read a second time.

This enabled following attack: A sufficiently malicious storage device might provide a backdoored file on the first read attempt, followed by the correct file on the second read attempt. The measurement would then appear correct. 

fixes #9
closed 2016-01-02 by neusdan