Rohde-Schwarz/TrustedGRUB2 #13: No event for PCR8/PCR9/PCR10/PCR11 in TPM event log

bsansoni, 2015-08-28:
I am currently using TrustedGrub2.
As I expected the PCR 8 (diskboot.S), PCR 9 (boot.S) , PCR10 and PCR11 are filled by the TrustedGrub2. I can see the value under my fedora 21 with the tpm_readpcr command.
In parallel, I used openpts for generation an attestation.
Openpts attestation is based on TPM eventlog.
So when I try to look at the TPM event log, no event are reported for PCR8,PCR9,PCR10,PCR11.
I check it out by typing the command "iml2text". (trace below).

Is it normal ? 
If yes is it planned to add this feature to TrustedGrub2 ?

Trace of iml2text.
[root@localhost TrustedGRUB2]# iml2text 
 Idx PCR       Type    Digest                                EventData
-----------------------------------------------------------------------
   0   0 0x00000008 c42fedad268200cb1d15f97841c344e79dae3320 [BIOS:EV_S_CRTM_VERSION]
   1   0 0x00000001 5553ddf5f17dbc8609fdce6a14ffc5be9faf2084 [BIOS:EV_POST_CODE(EV_CODE_NOCERT)]
   2   0 0x80000009 a50db81c18bf95bb16d5f375043818a926da8a0a [BIOS:EV_EFI_HANDOFF_TABLE,num=0x1,GUID=11d8ac35-fb8a-44d1-8d-09-0b-56-06-d3-21-b9]
   3   0 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
   4   1 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
   5   2 0x80000005 b8d3f0963299c8f016382bba508e56d260cbabb3 [BIOS:EV_EFI_RUNTIME_SERVICES_DRIVER,base=0xdc76e018,len=0x2e40,len=0x180000000,len=0x0]
   6   2 0x00000006 ab125862bd8dc852f1d2b5fad7eb8f2847f3943b [BIOS:EV_EVENT_TAG(EV_PLATFORM_SPECIFIC)]
   7   2 0x00000006 7d6f11c82dab57d0ed1b0c0d8a87954899b2b220 [BIOS:EV_EVENT_TAG(EV_PLATFORM_SPECIFIC)]
   8   2 0x00000006 868a1e48243629c8ba950faac0fdf0898d4405d5 [BIOS:EV_EVENT_TAG(EV_PLATFORM_SPECIFIC)]
   9   2 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  10   3 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  11   4 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  12   4 0x00000005 c1e25c3f6b0dc78d57296aa2870ca6f782ccf80f [BIOS:EV_ACTION, Calling INT 19h]
  13   4 0x00000005 38f30a0a967fcf2bfee1e3b2971de540115048c8 [BIOS:EV_ACTION, Returned INT 19h]
  14   4 0x00000005 410ec39247963e52793f7047bf2ac5f1eee25574 [BIOS:EV_ACTION, Booting Bcv Device P4: GLS85LS1032A CS 32GBN A101]
  15   4 0x0000000d 275d29c63c9b9a95f22be7284939a7990ead49ca [BIOS:EV_IPL]
  16   5 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  17   5 0x0000000e 33b2488503f7e1a56bda3a47b607e4d5ec30a51d [BIOS:EV_IPL_PARTITION_DATA]
  18   6 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  19   7 0x00000004 d9be6524a5f5047db5866813acf3277892a7a30a [BIOS:EV_SEPARATOR, ffffffff]
  20  10 0x00000000 b9e131be993a209f6a97f25559eb1c833bcd21fc [IMA:(TBD)]

bsansoni, 2015-08-28:
Additional information.
I used the tcglog grub command . And the PCR8 and PCR9 appears in the 'tcglog' trace.
Nevertheless the iml2text command does not report any PCR8 and PCR9 eventlog. 
Do you have another method to check the TPM event log?

bsansoni, 2015-08-28:
I found my issue.
iml2text fetch pcr through the TCSD (Trousers Daemon). 
The file /etc/tcsd.conf authorized only pcr 0,1,2,3,4,5,6,7 and 10. 
By adding 8,9,11 I can see the pcr 8,9.
Need to modify the iml2text to see the pcr 11.

Even if I found my problem it can help other. Not an issue from my side.

neusdan, 2015-09-01:
Currently only measurements of PCR 8 und 9 are logged. For the other measurements in PCR 10 - 12 TPM_Extend is used without additional log command. 

This is already on my todo list (#11)

Unfortunately im really busy at the moment with other stuff :(

neusdan, 2015-09-01:
Closing this now (duplicate #11)

closed 2015-09-01 by neusdan