Rohde-Schwarz/TrustedGRUB2 #15: UEFI feature

chava33, 2015-10-06:
I am not sure, where to ask a question, so just put it in the issue section..  Sorry

Thanks for the great work. Is there any work going on in enabling UEFI support or any estimated time?

neusdan, 2015-10-06:
I won't have time this year working on this. But maybe someone contributes it?!

seehwan, 2016-08-29:
I have strong interest on this subject. Could I join this work?
I have written some code, that actually works with my toy machine.
BTW. UEFI already have secureboot; do we need redundant implementation of boot measurement with TPM? Or can we say that they are complementary?

neusdan, 2016-09-06:
It would be nice if you could help with implementing UEFI support. Feel free to open some pull requests. I would prefer small incremental steps that are easy to review.

FYI: Matthew Garret has already implemented UEFI TPM support for GRUB2 here: 
https://github.com/mjg59/grub

Trusted boot and secure boot are probably redundant if only used to secure the boot process. But if you want for example to do remote attestation you still need the measured boot.

closed 2021-09-28 by securitykernel