Rohde-Schwarz/TrustedGRUB2 #16: TPM key exchange

chava33, 2015-10-06:
Hi Daniel Neus,

I am not sure, it is a good place to ask this question.. 

You have done the PCR read & write with trusted GRUB, just wondering, is it possible to extend that feature to retrieve keys from TPM as well (does TPM offers that API)?

I know with trousers, we can do it after boot but not sure we can do that during  BIOS/Boot loader stage.

Thanks

neusdan, 2015-10-07:
hi,

it is possible to execute/implement every possible TPM command that is listed in "TPM Main Specification Version 1.2 - Part 3 Commands".

So if it is possible with trousers it is also possible in bootloader stage. But unfortunately without Trusted Software Stack it isn't that easy/comfortable to implement.

chava33, 2015-10-07:
Thanks Daniel for the explanation.

closed 2015-10-07 by chava33