Rohde-Schwarz/TrustedGRUB2 #17: Booting should be possible even without TPM in the default configuration

neusdan, 2015-10-07:
But still keep a compile time flag to activate a "works with TPM only mode"

#14

momiji, 2015-11-02:
+1 I would like this option also.

While refactoring tpm-luks with a more user-centric approach, I utook the time to update to newest version 1.2.1, but it no more works on a virtual machine, because having a TPM is now mandatory.

The message I got is: No TPM found error, and nothing more at boot time.

I have no idea if it is better to have a compile options or a parameter somewhere in the grub.cfg or something else, even if I prefer configuration over compilation option, you choose :)

Anyway, great work! We're using 1.0.0 version in production, and we will move soon to 1.2.1 !!!

insecure, 2015-11-03:
If you decide to use TrustedGRUB instead of GRUB, you do this because you want to use the TPM. If you do not have a TPM, use traditional GRUB.
The reason for removing the "no-TPM fallback mode" was, that it is insecure:

1. block access to the TPM while GRUB tries to detect the TPM, e.g. by jamming the LPC bus
2. let GRUB boot your malicious OS (or even your malicious second stage)
3. unblock the TPM
4. extend some PCRs so they match a trusted platform configuration
5. Grats! You just upgraded your malicious OS to be a "trusted" OS. You may now unseal the key for the harddrive encryption and do anything else you like.

A security product should use secure defaults, so I would rather implement a compile flag which enables no-TPM mode. Anyway, as stated earlier, you would then probably prefer to choose common GRUB over TrustedGRUB.

neusdan, 2015-11-03:
Yeah you've convinced me. I think the other way round, i.e. compile flag which enables no-TPM mode, is the better way.

momiji, 2015-11-03:
that's ok for me too, no problem, having a flag will do the trick for me too...

savek-cc, 2016-10-06:
Is compiling TrustedGRUB2 to boot without TPM possible by now? (since this isn't closed yet)

neusdan, 2016-10-07:
no it's not implemented yet

jorgefm, 2018-02-23:
Any progress? Thanks!

jorgefm, 2018-02-23:
Well, this little patch does the trick...

--- a/grub-core/boot/i386/pc/boot.S	2017-06-08 22:17:01.000000000 +0200
+++ b/grub-core/boot/i386/pc/boot.S	2018-02-23 10:19:21.518294525 +0100
@@ -397,10 +397,12 @@
 
 error_tpm_deactivated:
   MSG(deactivatedTPM_error_string)
-  jmp	LOCAL(general_error)
+  popa
+  jmp	LOCAL(always_boot)
 error_no_tpm:
 	MSG(noTPM_error_string)
-	jmp	LOCAL(general_error)
+	popa
+	jmp	LOCAL(always_boot)
 
 /*
  * BIOS call "INT 1Ah, (AH)=BBh, (AL)=07h" TCG_CompactHashLogExtendEvent
@@ -450,6 +452,7 @@
 #endif /* ENABLE_TGRUB2 */
 /* END TCG Extension */
 
+LOCAL(always_boot):
 	/* boot kernel */
 	jmp	*(kernel_address)

jorgefm, 2018-02-23:
Well, this little patch does the trick...

`--- a/grub-core/boot/i386/pc/boot.S	2017-06-08 22:17:01.000000000 +0200
+++ b/grub-core/boot/i386/pc/boot.S	2018-02-23 10:19:21.518294525 +0100
@@ -397,10 +397,12 @@
 
 error_tpm_deactivated:
   MSG(deactivatedTPM_error_string)
-  jmp	LOCAL(general_error)
+  popa
+  jmp	LOCAL(always_boot)
 error_no_tpm:
 	MSG(noTPM_error_string)
-	jmp	LOCAL(general_error)
+	popa
+	jmp	LOCAL(always_boot)
 
 /*
  * BIOS call "INT 1Ah, (AH)=BBh, (AL)=07h" TCG_CompactHashLogExtendEvent
@@ -450,6 +452,7 @@
 #endif /* ENABLE_TGRUB2 */
 /* END TCG Extension */
 
+LOCAL(always_boot):
 	/* boot kernel */
 	jmp	*(kernel_address)
 `

jorgefm, 2018-02-23:
This little patch does the trick...

```
--- a/grub-core/boot/i386/pc/boot.S	2017-06-08 22:17:01.000000000 +0200
+++ b/grub-core/boot/i386/pc/boot.S	2018-02-23 10:19:21.518294525 +0100
@@ -397,10 +397,12 @@
 
 error_tpm_deactivated:
   MSG(deactivatedTPM_error_string)
-  jmp	LOCAL(general_error)
+  popa
+  jmp	LOCAL(always_boot)
 error_no_tpm:
 	MSG(noTPM_error_string)
-	jmp	LOCAL(general_error)
+	popa
+	jmp	LOCAL(always_boot)
 
 /*
  * BIOS call "INT 1Ah, (AH)=BBh, (AL)=07h" TCG_CompactHashLogExtendEvent
@@ -450,6 +452,7 @@
 #endif /* ENABLE_TGRUB2 */
 /* END TCG Extension */
 
+LOCAL(always_boot):
 	/* boot kernel */
 	jmp	*(kernel_address)
 ```

closed 2021-09-28 by securitykernel