Rohde-Schwarz/TrustedGRUB2 #19: How PCR usage is compliant with TCG ?

naynajain, 2015-10-12:
Hi,

Not sure if this is right place to ask this question, but I couldn't find any mailing list, so asking here.

I see that the first transition from pre-OS code to post-OS code, i.e. first boot sector of the diskboot.img image is itself extended to PCR[8] which is actually meant for static OS ? What is the reasoning for the way PCRs index are being used ?

Thanks & Regards,
          - Nayna

neusdan, 2015-10-12:
Hi, 

PCR 0-7 are filled by the BIOS. So i have used 8 upwards for my own measurements in TrustedGRUB2. As far as i know this should be compliant to the "TCG PC Specific Implementation Specification".

naynajain, 2015-10-13:
Who measures diskboot.img in your code ? As that is the first image to be loaded from grub2 ? So, is that not measured by BIOS ?
And is your implementation for TPM 1.2 or TPM 2.0 support ?
Also, is it in accordance with UEFI or BIOS ? I am assuming for BIOS.

neusdan, 2015-10-13:
> Who measures diskboot.img in your code

TrustedGRUB2 MBR bootcode -> boot.S

> As that is the first image to be loaded from grub2 ? So, is that not measured by BIOS ?

BIOS measures/loads TrustedGRUB2 MBR
TrustedGRUB2 MBR measures/loads diskboot.img
and so on

> And is your implementation for TPM 1.2 or TPM 2.0 support ?

For TPM 1.2

> Also, is it in accordance with UEFI or BIOS ? I am assuming for BIOS.

Only legacy BIOS measurements are supported for now.

All this information is already present in the readme

naynajain, 2015-10-13:
Thanks !! And which exact PCR index is used by BIOS to store TrustedGRUB2 MBR measurement ?

neusdan, 2015-10-13:
you should really have a look into the specification. It's PCR 4

naynajain, 2015-10-13:
Well, I guess my question wasn't clear enough. I understand from spec that it is PCR4, but wasn't sure what is actually done in TrustedGrub2 implementation, and so wanted to confirm which PCR did you choose for storing MBR measurements. 

Thanks :-)

neusdan, 2015-10-14:
The MBR is measured by the BIOS into PCR 4. I have no influence on that. The first measurement done by TrustedGRUB2 is in the MBR itself and stored into PCR 8 (diskboot.img)

closed 2015-12-28 by neusdan