Rohde-Schwarz/TrustedGRUB2 #20: Trustedgrub2 loading and going back to boot

naynajain, 2015-10-20:
I have installed trustedgrub2 successfully.
and now when i rebooted, it identifies the tpm, shows 
Booting From Harddisk
Trustedgrub2 loading....

And then goes reboot to "booting from hard disk"... It is just looping over this and not proceeding.
What might be the issue ?
Where can i check issues ?

neusdan, 2015-10-20:
You have to append `--no-rs-codes` to `grub-install` as explained in the README. I have to admit that this is easily overlooked. There is already an isssue for this here #18

naynajain, 2015-10-20:
So, for me after running grub-install command as:

grub-install --directory=/usr/lib/grub2/i386-pc --no-rs-codes /dev/sda2

I have chosen /dev/sda2, because my /boot/grub2 is on /dev/sda2,
it ha overwritten the MBR I guess.  So, when I rebooted, it was coming to grub prompt and wasn't recognizing trusted grub.

After that I executed grub-mkconfig -o /boot/grub2/grub.cfg and it has fixed the issue so trustedgrub2 started loading
But then it was looping back infinitely on TrustedGrub2 loading

So, anything wrong I did here ?

neusdan, 2015-10-20:
/usr/lib/grub2/i386-pc` is the directory where you have installed TrustedGRUB2?

To overwrite the MBR of the disk you have to use `/dev/sda` instead of `/dev/sda2`

`/boot` is automatically used for the additional TrustedGRUB2 files.

naynajain, 2015-10-20:
I actually tried both /usr/lib/grub2/i386-pc and /boot/grub2/i386-pc both... but nothing is working..
Now, even after mkconfig, it is not showing TrustedGrub2 loading... it is just going to grub prompt..

is there some entry to check for trusted grub ? I mean where is it installed and configured.

naynajain, 2015-10-20:
I specified only --root-directory and not --directory.. I think that correctly installed.
I did specify --no-rs-codes option also
But it is still looping around to Boot ESC menu and TrustedGRUB2 loading.

Is there any workaround so that I can come out of loop and proceed

neusdan, 2015-10-20:
i can't really follow in what you have done there. i suggest you install over normal grub2 and first try to install trustedgrub2 to a usb-drive for example.

chava33, 2015-10-20:
Try this, it worked for me from USB. 

Build and install
apt-get install autogen autoconf automake gcc bison flex
git clone Sirrix-AG/TrustedGRUB2

./autogen.sh

//Myself set INSTALLDIR=boot/trustedGrub2 and target x86_64 but it built for  i386 and boots fine
./configure --prefix=/boot/trustedGrub2 --target=i386 -with-platform=pc  
make
sudo make install

Install in USB (/dev/sdb)
 Bootable USB: With Gparted, USB is formated with Fat32 and Flag is set to boot & lba.
sudo mkdir /mnt/sdb1 
sudo mount /dev/sdb1 /mnt/sdb1 
sudo /boot/trustedGrub2/sbin/grub-install --directory=/boot/trustedGrub2/lib/grub/i386-pc --root-directory=/mnt/sdb1 --no-rs-codes /dev/sdb 
sudo cp /boot/initrd.img-3.16.0-46-generic /mnt/sdb1/boot/ 
sudo cp /boot/vmlinuz /mnt/sdb1/boot/ sudo umount /mnt/sdb1
Boot from USB (/dev/sdb)

grub> linux16 /boot/vmlinuz root=/dev/sda1
grub> initrd16 boot/initrd.img
grub> boot

closed 2015-12-30 by neusdan