Rohde-Schwarz/TrustedGRUB2 #23: Support for TPM2

wvervoorn, 2015-11-30:
The current implementation is TPM 1.2 based. Are there any plans to support TPM 2.0 in the future?

neusdan, 2015-12-01:
Would be nice to have support for TPM 2.0 but there are no concrete plans at the moment. Maybe in the next 1-2 years or less if someone contributes it.

wvervoorn, 2015-12-01:
Hello Daniel,

Thanks for your quick response. Unfortunately this is what I expected.

Best regards,

Wim Vervoorn

journey-wang, 2016-04-12:
I will

mikzaq, 2016-07-21:
Has there been any progress on this front?

neusdan, 2016-07-21:
Unfortunately no

mikzaq, 2016-07-21:
Do you know what would be required to add support for TPM 2.0? Also, are the components still hashed even if you boot without a TPM 1, and simply not stored in the appropriate PCRs, and if so, is there a way to output these values to a file instead?

neusdan, 2016-07-22:
> Do you know what would be required to add support for TPM 2.0?

No, i have to read the TPM 2.0 specification in order to answer this.

> Also, are the components still hashed even if you boot without a TPM 1, and simply not stored in the appropriate PCRs, and if so, is there a way to output these values to a file instead?

For TrustedGRUB2 a TPM is mandatory at the moment. Booting will fail if no TPM is found. Storing the hash values to a file doesn't seem like a good idea. I will not implemented something like this.

mikzaq, 2016-07-22:
@neusdan would you be willing/have time to add support for TPM 2.0? I would be happy to help, though I don't have much experience in this area so I would likely need guidance.

neusdan, 2016-07-28:
Sorry but in the near future I have no time to work on this.

Fluffy78, 2017-04-28:
Hi,

We are using TrustedGrubb2 and are our servers are now TPM2, so we need this capability. I don't have the time to deal with it, but if someone can add this functionality to the project, we could certainly look into financial compensation for adding this feature.

Thanks,

Cedric

wvervoorn, 2017-05-01:
Hello Cedric,

We can certainly add this functionality for you. We would need to discuss the financial compensation part offline.

If you are interested please contact me directly at : wvervoorn@eltan.com<mailto:wvervoorn@eltan.com>.

Best regards,

Wim

Fluffy78, 2017-05-01:
Hello,

Will reach out tomorrow, however just to clarify a point, the resulting work has to remain open source so that the community may benefit. If we could avoid forking, yet again, TrustedGrub, I think it would be most preferable for all.

Thanks

neusdan, 2017-05-01:
Fyi, Matthew Garret has implemented TPM 1.2 and TPM 2.0 support into GRUB2. He also submitted the patches upstream. But it looks like they are not integrated yet. You can read about it here:

https://lists.gnu.org/archive/html/grub-devel/2017-01/msg00029.html

If i understand correctly, his implementation is using the UEFI interface while TrustedGRUB2 uses the legacy BIOS interface.

Looks like the patches are also integrated into the CoreOS GRUB2:
https://github.com/coreos/grub

Fluffy78, 2017-05-04:
Hi,

My understanding is that, in general, the GNU policy is no dependency on TPM so it will never be mainstream, as such my alternatives are either tboot or TrustedGrub2. If you are using UEFI as a boot type, it means certification and at this point in time "RedHat" is mentionning "TrustedGrub" as being studied for release in 7.3+. until then the recommendation is to use "tboot" (hence no UEFI).

Therefore, I don't really see RedHat signing their boot loader with grub2.

FWIW, we are running our applications on RedHat Linux 7.

Thanks

neusdan, 2017-05-04:
> There's no problem including TPM support in GNU projects.

https://lists.gnu.org/archive/html/grub-devel/2017-01/msg00039.html

>  If you are using UEFI as a boot type, it means certification

If you are using Secure Boot you are right. But with Secure Boot disabled you can use unsigned bootloaders.

TheDash, 2017-05-09:
It's 2017 now. Should we all upgrade this to TPM 2.0? It's becoming very difficult to make and deploy production units without TPM 2.0

analysiser, 2017-06-22:
Thanks, I will try to take a look with gnu's fork and see what I can do. Sorry for the duplicate in #69

HaydonC, 2017-09-07:
Where is this at/is there currently a way to do a measured boot with a TPM2?

oba2cat3, 2018-01-29:
most of the patch is TPM related....

HaydonC, 2018-01-29:
As in, are you saying the patch is 'TPM1.2 related'? Just this thread is about TPM2 support and I couldn't see any mention of that in the patch

oba2cat3, 2018-01-31:
The patch was pulled from the coreos project, and focused on the code added by mjp59. AFAIK, it supports TPM2 functionality, as the coreos/grub supports TPM boot measure.

gu1234, 2018-02-14:
@HaydonC @oba2cat3 Did you manage to use TrustedGrub2 with TPM2?

oba2cat3, 2018-02-22:
@gu1234: nope - no need to. I use the grub fork from the coreos project. Specifically, the branch that was merged into TrustedGrub2 - https://github.com/suhho1993/TrustedGRUB2-1/commit/e1b2b265afd448229e298f862305a8dfdfc7ad6c
works as expected, just make sure you "mark hold" the Grub version after you install, or the OS will try to update it later with unexpected results....

gu1234, 2018-02-22:
Thanks @oba2cat3

HaydonC, 2018-02-27:
@gu1234  did you get this to work? 
@oba2cat3 I'm having issues installing the 'e1b2b26' branch. Are you saying you currently have this branch running with TPM2.0? If so I'll persist with it!
Thanks for the help guys

HaydonC, 2018-02-27:
@oba2cat3 Have now installed the 'e1b2b26' commit. However, although it doesn't seem to register anything to a TPM2? Only pcr's 0-7 are filled by the BIOS and 8&9 are left as zero's

gu1234, 2018-02-27:
@HaydonC I didn't get to try installing it yet, would appreciate if you could let us know if you were able to get all the PCRs filled

According the trustedgrub2 documentation it should fill up all the PCRs up to 13

HaydonC, 2018-02-27:
@gu1234  Using this branch with TPM1.2 I get PCRs 0-10 filled. A fresh install of Ubuntu will fill 0-7 (from the BIOS) and 10 of the PCRs.
Using TPM2.0 and this branch I still only get 0-7 filled. 
Where did you read that up to 13 will be filled? Was it specific to TPM2.0?

gu1234, 2018-02-27:
@HaydonC It's in the TrustedGrub readme: https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2#13-measurements-in-short

Although maybe it needs to be configured as specified here: https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2#161-pcr-selection

HaydonC, 2018-03-05:
@gu1234 Yeah, those are definitely correct for TPM1.2 but I think not for TPM2.0. Need to get @oba2cat3 to confirm whether they're getting just 0-7 pcrs filled (from BIOS) or whether TrustedGRUB is actually writing to the TPM2 as well

oba2cat3, 2018-03-06:
@HaydonC - it is very hard to help you from my perspective. My Measured/Trusted boot solution is a handcrafted one, with a personally patched Coreos/grub in the middle of it. @mjg59 *original* solution is documented here (on his blog): https://mjg59.dreamwidth.org/37656.html
but his final code (implemented in the coreos/grub) is not that extreme, and AFAIK reuses the same PCR slot for all the measurements done by Grub. I think it was PCR8.
reminder - TGrub2 was a solution I decided not to test/use exactly because there was no TPM2 support at the time.

mmelamud01, 2018-03-26:
Hi,

I installed trustedgrub2 from suhho1993 branch that supposed to support TPM2 , i followed the instructions found here : https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2#13-measurements-in-short

I also change the BIOS from UEFI to Compatibility Support Module (CSM) that provides legacy BIOS compatibility.

But still when booting and entering the GRUB selection menu , it is still appears as GRUB2 and not trustedgrub.

Any ideas?

Thanks,
Michael

mjg59, 2018-03-26:
@mmelamud01 There is no way to use TPM2 from CSM. You must use native UEFI booting.

mmelamud01, 2018-03-27:
@mjg59  - But then in this case trustedgrub2 wont work as it expects legacy BIOS ,or i am mistaken?

mjg59, 2018-03-27:
@mmelamud01 I don't know what tree you're running, but there's no BIOS interface specified for TPM2. The only x86 firmware interface for TPM2 requires UEFI.

NeryHenrique, 2018-04-02:
Hi guys, I have actually a similar problem. I need to use TPM2 (or PTT) to save an key for LUKS. 
Before, I could use thrusted grub with tpm1.2 all everything works as expected. But now I don't know whats the best solution.. what do you think?
I was thinking in try UEFI with this trusted grub patch and use tpm_2.0_tss as driver. All in Ubuntu.

HaydonC, 2018-04-04:
Hi all, I was in a similar situation to @NeryHenrique and have recently resolved the issue. Using a TPM2 with UEFI boot and the master branch of the coreos grub ( https://github.com/coreos/grub/tree/master )  you can get PCRs 0-9 registered. 0-7 are measurements from UEFI whilst 8-9 come from GRUB2 and measure initramfs/kernel/OS. 
You can follow https://robertou.com/tpm2-sealed-luks-encryption-keys.html for a version of the LUKS implementation mentioned previously

NeryHenrique, 2018-04-06:
Oh good to know! Did you do it in Ubuntu? What tools did you use? tpm_tis, trousers, luks, tpm_tools? 
Did anyone already used Intel firmware tpm (tpp) is it indiferent from using the tpm2.0?

HaydonC, 2018-04-09:
The tool set I've used is the IBM TSS - https://sourceforge.net/projects/ibmtpm20tss/

NeryHenrique, 2018-04-10:
Nice, thanks. i ll try to use it!

jorgefm, 2018-07-23:
@HaydonC - Could you help me using TPM2 with UEFI? I've downloaded the coreos and compiled ok. Could you detail the step you made to install the grub in your hard disk? The files copied and where.
Thanks!

jorgefm, 2018-07-24:
What I've done is download the grub2 from the coreos grub master and compile it with:
./configure --with-platform=efi --target=x86_64 --disable-werror

I have a boot partition with FAT format in sde1 and the system partition in sde2. In sde1 I have the directories /boot and /EFI/BOOT. I've installed grub2 with:

# mkdir /tmp/kk
# mount /dev/sde1 /tmp/kk
# /sbin/grub-install --removable --target=x86_64-efi --boot-directory=/tmp/kk/boot --efi-directory=/tmp/kk/EFI/BOOT /dev/sde
# umount /tmp/kk

My grub.cfg is a very simple one:

menuentry 'test' {
    linux (hd0,2)/vmlinuz
    initrd (hd0,2)/initrd
}

My system boots but I only get the PCR 0 to 7 some values. The PCR8 to 16 always are zero.

How do you force to extend the PCRs? I'm using a SBL9665 (TPM2.0)

Thanks!

heraklit256, 2018-08-06:
I would be also interested in this.

rhopfer, 2018-08-09:
I installed the fork of CoreOS (https://github.com/coreos/grub), main branch 2.02-coreos and it works for me. I think the master branch ist the upstream without TPM.
The most important difference to TrustedGrub2 is, grub-coreos measures all commands to PCR[8] and all binaries to PCR[9].

jorgefm, 2018-08-16:
Hi Roland,
I'm using the CoreOS fork and another difference is that TPM is only implemented for UEFI. With legacy you don't get TPM support... or I'm not been able to get it :)

zaolin, 2018-08-16:
This is incorrect. You can use coreboot + seabios which gives you the same capabilities for TPM 1.2 and 2.0.

jorgefm, 2018-08-16:
As Zaolin points I've been able to use TPM1.2 and 2.0 under LPC with coreboot + seabios, but @mjg59 is the top expert then I believe him :)

zaolin, 2018-08-16:
Oo https://git.seabios.org/cgit/seabios.git/tree/src/tcgbios.c

zaolin, 2018-08-16:
The interface supports TPM 1.2 and 2.0, see pass_through_to_tpm_int

mjg59, 2018-08-21:
Yes, SeaBIOS has extended the TCG interface to also operate on TPM 2 devices - that's beyond the spec, though, and is unlikely to work on any other BIOS implementation. If you're targeting a SeaBIOS-based system then that's a reasonable approach to take.

irn73, 2018-10-11:
Hi jorgefm & Roland,

I also tried to the coreos grub2 with TPM2.0 on UEFI system and also got PCRs 8 to 16 always zero.
Did someone here manged to accomplish  measurements with grub2 on PCRs 8 to 16?
Can you please privide some instructions how to do it?

Thanks!

irn73, 2018-10-14:
Hello Roland,

Thank you for your quick prompt.
I also tried your branch but still I don't get any measurements.
I have an EFI boot partition on /dev/sda:

`fdisk -l /dev/sda`
`WARNING: fdisk GPT support is currently new, and therefore in an experimental phase. Use at your own discretion.`

`Disk /dev/sda: 128.0 GB, 128035676160 bytes, 250069680 sectors`
`Units = sectors of 1 * 512 = 512 bytes`
`Sector size (logical/physical): 512 bytes / 512 bytes`
`I/O size (minimum/optimal): 512 bytes / 512 bytes`
`Disk label type: gpt`
`Disk identifier: 7237D4D7-15A9-47A0-BC19-286F861979A2`

`#         Start          End    Size  Type            Name`
` 1         2048       411647    200M  EFI System      EFI System Partition`
` 2       411648      2508799      1G  Microsoft basic`
` 3      2508800    250068991    118G  Linux LVM`

I have ran: 

`/usr/local/sbin/grub-install --target=i386 --directory=/usr/local/lib/grub/i386-pc/  --force /dev/sda`

and reboot the system

My system boots but only PCR 0 to 7 get values. PCRs 8 and above are zero.

Am I doing something wrong here?

Thanks,
Ido

irn73, 2018-10-16:
I'm trying to run this over CentOS but it doesn't seems to work.
The file grubx64.efi is regenerated and the grub doesn't load.

Is it possible to run the grub-install over CentOS?

Thanks

rhopfer, 2018-10-16:
And you have mounted your EFI partition to /boot/efi?
There is no reason why it should not work with CentOS.

irn73, 2018-10-17:
My EFI partition is indeed mounted:

> /dev/sda2               1014M  152M  863M  15% /boot
/dev/sda1                200M   10M  190M   5% /boot/efi

The problem is that the outcome of grub-install is a new grubx64.efi file which is not found during the boot process. The error is: "Failed to open \EFI\BOOT\grubx64.efi - Not Found".
It seems that after running the grub-install the path for searching the grubx64.efi file has changed.

I also tried to copy it to /boot/efi/EFI/BOOT folder. The outcome is that the Linux has come up, but with the old grub (the original one) and hence didn't activate the measurement inside the TPM.

pietrushnic, 2019-09-07:
@zaolin I assume SeaBIOS implementation use INT 1Ah API. Are you aware of any OS/bootloader (except TrusedGRUB2) that leverages further SeaBIOS IRQ  to get fully measured boot? 

It looks like TrustedGRUB2 cannot leverage TPM2.0 support from SeaBIOS. 

@mjg59 I would like to understand how INT 1Ah defined in [TCG PC Client Specific Implementation Specification for Conventional BIOS](https://www.trustedcomputinggroup.org/wp-content/uploads/TCG_PCClientImplementation_1-21_1_00.pdf) is beyond the spec, or maybe you just mean that for mentioned spec TPM2.0 is not supported?

@wvervoorn @Fluffy78 did you managed to move the implementation of TPM2.0 support further?

3mdeb will present [Non-UEFI-aware measured boot using coreboot, GRUB and TPM2.0](https://linuxplumbersconf.org/event/4/contributions/517/) at LPC SystemBoot MC. If you have any points that we should rise during discussion please let me know.

securitykernel, 2021-09-28:
Thanks to everyone contributing here. We eventually decided to deprecate and no longer maintain this project. I will be closing this issue.

closed 2021-09-28 by securitykernel