Rohde-Schwarz/TrustedGRUB2 #24: Question: Is it possible to use TrustedGRUB2 with tpm-luks to get keys that don't rely on good state?

mtahmed, 2015-12-11:
I want to be able to decrypt the rootfs from TrustedGRUB2 using keys from TPM that were not sealed (do not rely on a good state). Is that possible? Is there an example config I can use to achieve that?

Thanks!

neusdan, 2015-12-11:
Thats not possible at the moment. TrustedGRUB2 can only use sealed LUKS keyfiles when using the `cryptomount` command.

So if i understand you correctly you wan't to read out a TPM key and pass this to LUKS to decrypt the volume?

mtahmed, 2015-12-11:
Correct. I want to be able to read out a TPM key, pass it to maybe the cryptomount command, which unlocks the LUKS-encrypted volume for me.

mtahmed, 2015-12-11:
Do you know how feasible/easy it would be to add that functionality? I am willing to do the work and send a pull request if I could get some pointers.

neusdan, 2015-12-12:
To answer this question i need more information how tpm-luks works. Which key is used to encrypt the data? As far as i know it is not possible to read out TPM keys in the clear. You can only send a encrypted data blob to the TPM to either unseal or unbind the data. In the latter case the data is only bind to the TPM and not additionally to the plattform configuration.

So i assume tpm-luks generates a key and encrypts this key with a TPM key. But you have to look up this.

mtahmed, 2015-12-14:
Well, `tpm-luks` is able to get the keys from the TPM NVRAM. It uses `nv_readvalue` to read the NVRAM at some given offset. I was able to achieve that using tpm-luks already and thanks for the help. 

I do have a question about what values to pre-compute when I update the initrd and linux and I would appreciate any help with that. I will ask that in a different question.

closed 2015-12-14 by mtahmed