Rohde-Schwarz/TrustedGRUB2 #35: Multiboot files not measured

mtahmed, 2016-01-05:
Kernels loaded using the `multiboot` command don't seem to be measured. E.g. a typical Xen menuentry looks like this:

```
menuentry 'Xen with Linux 3.2 ...' {
  ...
  multiboot /xen.gz placeholder ${xen_rm_opts}
  module /vmlinuz-3.2 placeholder root=/dev/sda1 ...
  module --nounzip /initramfs-3.2.img
  ...
}
```

In this case, the files `xen.gz`, `vliminux-3.2`, `initramfs-3.2.img` need to be measured but aren't. The command functions are defined in `grub-core/loader/i386/xen.c`.

The change seems simple enough if I don't follow the new convention of measuring the same buffer that is loaded into memory. Would you be able to pick this up or would you rather I work on it?

neusdan, 2016-01-05:
Thanks for reporting. I'll look into this the next days.

neusdan, 2016-01-06:
Could you upload those files or tell me a location where i can download some multiboot files for testing?

mtahmed, 2016-01-07:
If you install Xen on your distribution, it will provide those files and install the relevant grub menu entries. I have also attached the `xen.gz` for use with the multiboot commands and whatever the local kernel/initrd are can be used for `module`
[xen.gz](https://github.com/Sirrix-AG/TrustedGRUB2/files/81635/xen.gz)
 commands.

neusdan, 2016-01-07:
Could try the implementation from PR #37 ?

mtahmed, 2016-01-09:
Tried and it works as expected. Thanks a lot for doing this!

PR #37 (2016-01-09): implemented multiboot measurement
Loader measurements for `multiboot` and `module` commands

Unfortunately the `multiboot` measurement uses the deprecated `measure_file` method which does not protect against malicious storage devices ( GH #9 ). I've tried to implement the `measure_buffer` method but it's a lot more work and maybe even requires grub2 api changes.

In any case i think it's really hard to exploit. But pull requests are welcome. I'll create an issue for that.

Closes #35
closed 2016-01-09 by neusdan