Rohde-Schwarz/TrustedGRUB2 #43: Tested TPM modules/IC

ronaldonunez, 2016-06-30:
What is the list of TPM modules/IC compatible with the TrustedGRUB2? I successfully tested it with Infineon TPM IC (present in most of TPM modules). I would like to know if Nuvoton NPCT620A TPM IC also compatible with it?

neusdan, 2016-07-04:
There is no such list. Unfortunately i can't tell if this TPM is supported or not because i have no such TPM here.

zaolin, 2016-07-04:
The TrustedGrub 2 uses the firmware interface to call the TPM. Normally all TPM's supported by the firmware itself should work!

zaolin, 2016-07-04:
It can happen that some commands are incompatible with TPM itself but if so something is wrong on the TPM side. TrustedGrub 2 only supports TPM v1.2 devices. The TPM you posted is not listed on the company homepage ?!

ronaldonunez, 2016-07-25:
Thanks for your answers. I was not able to find any information about NPCT 620A on Nuvoton web site. Based in what I found here: http://permalink.gmane.org/gmane.comp.encryption.tpmdd.devel/1198  I can believe that the NPCT6xx is a TPM2.0 IC, which is not supported by TrustedGrub2.

closed 2016-07-25 by ronaldonunez