Rohde-Schwarz/TrustedGRUB2 #45: TrustedGRUB2 fails to boot with TPM 1.2 on a HP desktop

amitlimaye, 2016-07-13:
Hello i compiled the grub source from here and tried loading on my HP Z440 desktop. 
I have enabled the TPM security knob in my BIOS but i am seeing this error and my system refuses to Boot 

+++++++++++++++++++++++++
rustedGRUB2 loading 
Welcome to TrustedGRUB2!
TCG_HasgLogExtendEvent failed : 0x2 
Aborted. Press any key to exit

+++++++++++++++++++++=
Any ideas what am i doing wrong here. 
I looked the TCG PC spec the error number 0x2 seems to indicate that i got TCG_PC_LOGOVERFLOW 
it means insufficient memory to create log entry. What am i missing here 
my dmesg output when i boot without trusted grub shows this 

tpm_tis 00:09: 1.2 TPM (device-id 0x1A, rev-id 16)

neusdan, 2016-07-14:
Strange. Never seen this before. Probably a bios bug. Are you running the latest version?

amitlimaye, 2016-07-14:
I got the source from the git tree. 
I was thinking BIOS bug as well i am going to try the same thing on a different vendor server. 
The  TCG_PC_LOGOVERFLOW  made me think if i needed to configure something in my BIOS or TPM chip like carving out a region for logs in NVRAM before i try to boot with this. The BIOS has no further options beyond enabling and disabling the chip. (or making it hidden). All those things are set as HP asks me to

neusdan, 2016-07-14:
Do you have the latest BIOS version installed?

amitlimaye, 2016-07-14:
havent checked but i got this desktop from HP last week
will reach out to HP to check it

-SIGTERM
amit

Humanity's first sin was faith, Its first virtue doubt

amitlimaye, 2016-07-14:
looks like my BIOS is a few versions old i have 2.14 the latest is 2.19

white00black, 2016-07-29:
I have a similar issue, it does not work with version 1.4 and master. Try version 1.3, should work.

neusdan, 2016-07-29:
Could you tell us what error exactly?

johnwallace123, 2016-11-04:
@amitlimaye @white00black: Can you try the code from PR #55 and see if that resolves your issue?  It looks like there's a widespread bug in newer BIOSes that prevents appending to the TPM event log with certain calls.

closed 2021-09-28 by securitykernel