Rohde-Schwarz/TrustedGRUB2 #52: TPM Error on boot

ernvalentino88, 2016-09-09:
Hi,
I've installed TrustedGRUB2 on an ubuntu 14.04 machine with a TPM v1.2. The installation was completed without errors, but when I reboot the PC this error while booting was show. What this error means? Thank you 

Ernesto

neusdan, 2016-09-14:
What exact error code / message is shown?

ernvalentino88, 2016-09-14:
Hi,
thanks for answering. Unfortunately the only message shown on the screen is:
TrustedGRUB2 Loading
Tpm Error

and nothing else, that's why I asked...

johnwallace123, 2016-10-07:
Have you enabled the TPM in the BIOS?  If the TPM is not available, trusted grub won't boot, and you'll get an error similar to that.  Note that in most BIOSes, the TPM is not enabled in the factory default settings.

ernvalentino88, 2016-10-07:
TPM is enabled and I can use it.

neusdan, 2016-10-07:
Hard to tell what's the problem here without even an error code. TPM hardware problem, TPM firmware problem or a BIOS bug for example.

johnwallace123, 2016-10-07:
When booting without TrustedGRUB, can you see the PCRs and the measurement log of the TPM?

Assuming RHEL7 or derivative, the PCRs are located at /sys/class/tpm/tpm0/device/pcrs and the measurement log can be found at /sys/kernel/security/tpm0/ascii_bios_measurements.  Without TrustedGRUB, you should still see PCRs 0-7 populated with nonzero values and those corresponding measurements in the log.

Variables that are also of interest are /sys/class/tpm/tpm0/device/enabled and (same path)/temp_deactivated.

ernvalentino88, 2016-10-07:
Yes, I can see first 8 PCRs not filled with 0 and also IMA is working. I don' think the error is in the TPM, probably in the BIOS

johnwallace123, 2016-10-11:
Can you check to see if PR #54 solves your issue?  It seems that there's a recent, issue across many mainboard manufacturers that may cause a problem with TrustedGRUB2.  The symptoms have been varied, but the solution seems to be universal.

closed 2021-09-28 by securitykernel