Rohde-Schwarz/TrustedGRUB2 #56: Possible issue

hunter86bg, 2016-10-25:
Hi guys,

as I'm new to TrustedGRUB 2 , I'm not sure that the situation that have happened to me is really an issue.
Let me summarize.
1. I have 1 encrypted with LUKS drive.
2. I have enabled the TPM and started all relevant services
3. Compiled the TrustedGRUB 2 (there was a warning regarding "locales" , I think)
4. Booting failed (probably didn't find stage 1.5 or stage 2 )

I'm trying to make the TPM unlock the LUKS encrypted partition without user intervention.

Please provide your guidance ,as I expected that TrusedGRUB 2 should still boot the system and LUKS to ask for password.

neusdan, 2016-10-25:
What do you mean with "Booting failed" ? What's the exact error message?

hunter86bg, 2016-10-25:
Hah, Grub Rescue console.

neusdan, 2016-10-26:
I assume something went wrong during installation

hunter86bg, 2016-10-26:
I tried it again and it was the same issue. When I compared my grub2 backup folder and the new "grub" folder I have noticed that there is no grub.cfg and no device.map . I 've copied them and now it boots normally.

neusdan, 2016-10-26:
So this issue can be closed?

hunter86bg, 2016-10-26:
It should be checked why it didn't create the grub.cfg and the device.map files. A software is working , when there is no need for workarounds.
Do you agree with me ? If not, maybe the guide should be updated...

neusdan, 2016-10-26:
As far as i know the `grub-install` command does not create a grub.cfg file, so no surprise here. There is actually a warning in the TrustedGRUB2 README:

> [WARNING] if installing over an old GRUB2 install you probably have to adjust your grub.cfg

hunter86bg, 2016-10-27:
I missed that.It seems there is no issue at all. 
Thanks for the answers.

closed 2016-10-27 by hunter86bg