Rohde-Schwarz/TrustedGRUB2 #60: Unable to install on MBR

stratoss, 2016-12-05:
Hi,

The normal grub2 is installing and running properly, but when I try to install TrustedGRUB2 I receive:

> sudo /opt/trustedgrub2/sbin/grub-install --directory=/opt/trustedgrub2/lib/grub/i386-pc/ /dev/sda
Installing for i386-pc platform.
/opt/trustedgrub2/sbin/grub-install: warning: cannot open directory `/opt/trustedgrub2/share/locale': No such file or directory.
/opt/trustedgrub2/sbin/grub-install: error: disk `lvm/sda1_crypt' not found.

sda1_crypt is the alias for the decrypted /dev/sda1 which is my boot partition.

neusdan, 2017-06-11:
If this is still relevant for you, can you check if current master is still throwing this error?

stratoss, 2017-06-12:
Will check and get back to you in 1-2 days. Thanks.

tahirahmad1985, 2017-11-28:
Hi stratoss, could you please help me and put some comments on issue 70

tahirahmad1985, 2017-11-28:
Sorry, actually issue 72

closed 2021-09-28 by securitykernel