Rohde-Schwarz/TrustedGRUB2 #61: TrustedGRUB2 hangs without errors

fchiacchiaretta, 2017-03-01:
Hi,
I built trustedgrub2 and installed it, no error when installing it on /dev/sda.
TPM is enabled but when I reboot my server after installation (Lenovo X3650 M5), I get stuck at:

```
TrustedGRUB2 loading.
Welcome to TrustedGRUB2!
```

/boot partition in not encrypted.
No errors are thrown even after waiting several minutes.

Is there a way to troubleshoot this issue?

fchiacchiaretta, 2017-03-01:
I managed to compile with debug enabled, on reboot I get the following message:

```
TrustedGRUB2 loading.
Welcome to TrustedGRUB2!

measured module: terminal
SHA1 of buffer: _40-digits string_
```

neusdan, 2017-03-01:
Seems to be related to #53. The changes in #55 probably solves this problem.

fchiacchiaretta, 2017-03-02:
Thanks for pointing it out. Following [#55](https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2/pull/55) and compiling with TGRUB_NOEVENTLOG flag the boot process goes on as expected.

Thanks for your support.
Should I close this issue or leave it open until [#55](https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2/pull/55) is merged?

neusdan, 2017-03-02:
Glad this works for you. You can leave it open.

closed 2021-09-28 by securitykernel