Rohde-Schwarz/TrustedGRUB2 #62: TCG_PassThroughFail: 0xc0000 occurs in unseal procedure w TPM1.2

rootae, 2017-03-07:
What I am doing is build TrustedGrub2 on Ubuntu 16.04 in Broadwell DE / Xeon platform (AMI BIOS:core version 5.0.11 0.20 x64).
Nonetheless, after using the tpm-tool to seal data with PCRS 8 & 9, the TrustedGrub2 was unable to unseal the with the key. 
Is there any specific mismatch in seal/unseal procedure or detail information about the return error code (TCG_PassThroughFail: 0xc0000)

Thank you!

neusdan, 2017-03-14:
Error codes are provided in the TPM specification. 

In order to help i need more information what you are doing. What TrustedGRUB2 commands are you using?

You can also try to turn on debug mode:
https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2#162-debug-output

tahirahmad1985, 2017-03-29:
Hi, 

let me clear you what exactly we are trying to do,

The general idea of this test case is the following: We will seal a piece of data using TPM and try to unlock it from TrustedGrub2. Every time I try to unseal the key I get a TCG_PassThroughFail: 0xc0000 error.

The test is as follows:

Download VirtualBox or VMWare

Download Ubuntu 16.04 (http://releases.ubuntu.com/16.04/ubuntu-16.04.2-desktop-amd64.iso)

Install the OS

sudo apt-get update

sudo apt-get install build-essential automake autopoint libtool libtspi-dev bison flex git

wget https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2/archive/1.4.0.tar.gz

tar xzf 1.4.0.tar.gz

cd TrustedGRUB2-1.4.0/

export INSTALL_DIR=/path/to/install_dir

./autogen.sh

./configure --prefix=$INSTALL_DIR --target=i386 -with-platform=pc

make CPPFLAGS=-DTGRUB_DEBUG && make install

If everything goes well we should have a file called grub-install under $INSTALL_DIR/sbin/

I installed TrustedGrub2 into a USB stick using the following command:

sudo $INSTALL_DIR/sbin/grub-install --directory=$INSTALL_DIR/lib/grub/i386-pc /dev/sdb ; # device name may be different in your case

Build tpm-tools and tpm ownership

I used Ubuntu 16.04 running on a machine that we want to seal/unseal a key.

Clear and enable TPM (from BIOS)

Boot the OS

sudo apt-get update

sudo apt-get install git

git clone https://github.com/shpedoikal/tpm-tools.git

git checkout tpm-sealdata-raw (# checking out this branch is very important because it adds the -r option to tpm_sealdata).

sudo apt-get install automake autoconf libtool gettext trousers trousers-devel libtspi-dev autopoint (link to instructions https://github.com/shpedoikal/tpm-tools)

sh ./bootstrap.sh

export TPM_DIR=/path/to/tpm_build_dir

./configure --prefix=$TPM_DIR

make && make install

$TPM_DIR/sbin/tpm_takeownership -y -z

$TPM_DIR/sbin/tpm_setenabled --enable -z

$TPM_DIR/sbin/tpm_setactive -z

Create a key

echo “TPM UNSEAL FROM GRUB” > /tmp/key

#seal the key now that we own the TPM using PCRs 8 and 9

$TPM_DIR/bin/tpm_sealdata -p 8 -p 9 -z -r -i /tmp/key -o /tmp/key.enc

If everything goes well we should have a sealed key named ‘/tmp/key.enc’.

We should now copy the sealed key to a place that we can access from TrustedGrub2. I copied /tmp/key.enc to the root of the USB drive that we installed TrustedGrub2.

Reboot the system and boot it using the newly created image. Press ‘c’ in the TrustedGrub2 menu. Execute the following command from the grub menu:

grub> unseal /root/key.enc

I always get an error after this step
TCG_PassThroughFail: 0xc0000

closed 2021-09-28 by securitykernel