Rohde-Schwarz/TrustedGRUB2 #64: TCG_PassThroughFail: 0xc0000 while unsealing the key

tahirahmad1985, 2017-03-22:
The general idea of this test case is the following: We will seal a piece of data using TPM and try to unlock it from TrustedGrub2. Every time I try to unseal the key I get a TCG_PassThroughFail: 0xc0000 error.

 
The test is as follows:

 

1.       Download VirtualBox or VMWare

2.       Download Ubuntu 16.04 (http://releases.ubuntu.com/16.04/ubuntu-16.04.2-desktop-amd64.iso)

3.       Install the OS

4.       sudo apt-get update

5.       sudo apt-get install build-essential automake autopoint libtool libtspi-dev bison flex git

6.       wget https://github.com/Rohde-Schwarz-Cybersecurity/TrustedGRUB2/archive/1.4.0.tar.gz

7.       tar xzf 1.4.0.tar.gz

8.       cd TrustedGRUB2-1.4.0/

9.       export INSTALL_DIR=/path/to/install_dir

10.   ./autogen.sh

11.   ./configure --prefix=$INSTALL_DIR --target=i386 -with-platform=pc

12.   make CPPFLAGS=-DTGRUB_DEBUG && make install

 

If everything goes well we should have a file called grub-install under $INSTALL_DIR/sbin/

I installed TrustedGrub2 into a USB stick using the following command:

sudo $INSTALL_DIR/sbin/grub-install --directory=$INSTALL_DIR/lib/grub/i386-pc /dev/sdb ; # device name may be different in your case

 

 

Build tpm-tools and tpm ownership

 

I used Ubuntu 16.04 running on a machine that we want to seal/unseal a key.

1.       Clear and enable TPM (from BIOS. I replicated this problem using both v0.08 andv 0.13).

2.       Boot the OS

3.       sudo apt-get update

4.       sudo apt-get install git

5.       git clone https://github.com/shpedoikal/tpm-tools.git

6.       git checkout tpm-sealdata-raw (# checking out this branch is very important because it adds the -r option to tpm_sealdata).

7.       sudo apt-get install automake autoconf libtool gettext trousers trousers-devel libtspi-dev autopoint (link to instructions https://github.com/shpedoikal/tpm-tools)

8.       sh  ./bootstrap.sh

9.       export TPM_DIR=/path/to/tpm_build_dir

10.   ./configure --prefix=$TPM_DIR

11.   make && make install

12.   $TPM_DIR/sbin/tpm_takeownership -y -z

13.   $TPM_DIR/sbin/tpm_setenabled --enable -z

14.   $TPM_DIR/sbin/tpm_setactive -z

15.   # Create a key

16.   echo “TPM UNSEAL FROM GRUB” > /tmp/key

17.   #seal the key now that we own the TPM using PCRs 8 and 9

18.   $TPM_DIR/bin/tpm_sealdata -p 8 -p 9 -z -r -i /tmp/key -o /tmp/key.enc

 

If everything goes well we should have a sealed key named ‘/tmp/key.enc’.

 

We should now copy the sealed key to a place that we can access from TrustedGrub2. I copied /tmp/key.enc to the root of the USB drive that we installed TrustedGrub2.

 

Reboot the system and boot it using the newly created image. Press ‘c’ in the TrustedGrub2 menu. Execute the following command from the grub menu:

grub>  unseal /root/key.enc

 

I always get an error after this step: TCG_PassThroughFail: 0xc0000

albal, 2017-05-05:
I don't think TPM is supported in a VM.

zaolin, 2017-05-05:
It is and it works normally. But it looks like that SeaBIOS > 1.9.3 is broken for TPM 1.2 which known to be used in QEMU

closed 2017-11-28 by tahirahmad1985