Rohde-Schwarz/TrustedGRUB2 #66: Error build TrustedGrub2

Body4, 2017-05-04:
Hello,

I'm installing my new PC on archlinux and I have an error when I try to compile TrustedGRUB2.
I don't know if my problem is related with TrustedGrub so I post my issue here. But I'm sorry in advance if I'm wrong.

When I try to compile TrustedGRUB2, I have this error:
```
grub-core/osdep/unix/getroot.c: Dans la fonction « grub_util_biosdisk_is_floppy »:
grub-core/osdep/unix/getroot.c:751:13: erreur : In the GNU C Library, "major" is defined
 by <sys/sysmacros.h>. For historical compatibility, it is
 currently defined by <sys/types.h> as well, but we plan to
 remove this soon. To use "major", include <sys/sysmacros.h>
 directly. If you did not intend to use a system-defined macro
 "major", you should undefine it after including <sys/types.h>. [-Werror]
   if (major(st.st_rdev) == FLOPPY_MAJOR)
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                            
cc1 : tous les avertissements sont traités comme des erreurs
make[2]: *** [Makefile:5618: grub-core/osdep/unix/libgrubkern_a-getroot.o] Error 1
make[1]: *** [Makefile:10703: all-recursive] Error 1
make: *** [Makefile:3099: all] Error 2
```
Here is my version of gcc:
```
Utilisation des specs internes.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/6.3.1/lto-wrapper
Cible : x86_64-pc-linux-gnu
Configuré avec: /build/gcc-multilib/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release
Modèle de thread: posix
gcc version 6.3.1 20170306 (GCC) 
```
I try to add "#include <sys/sysmacros.h>" but I have another smiliar error with "makedev".

Thank in advance for your help.

neusdan, 2017-06-11:
You can please check current master if still is still an issue?

max-k, 2018-01-02:
I confirm this is still an issue.
TrustedGRUB2's master branch don't build with GCC 7.2.1.

closed 2021-09-28 by securitykernel