Rohde-Schwarz/TrustedGRUB2 #73: Backporting old options

jorgefm, 2018-02-09:
Hi,
When I was using grub-0.97 (I know... long time ago) there were some very useful options for embedded systems to avoid a lot of files in the physical support, facilitating the update process for instance. With this version I only have three files: stage1, e2fs_stage1_5 and stage2. That's perfect for rigid environments :)

Now I need some TPM support and I need to update my boot process. I'm missing two options and I would like to ask how difficult could be implementing them.

1) In the old version was a configure option --enable-preset-menu to embed a grub.cfg file inside the stage2 file.
2) In the new version there are a lot of modules files in the i386-pc directory. A full of them are not needed in my system. Is it possible to configure the compiling process to customize which modules to build? and is it possible to avoid modules and generate a monolithic file (the previous stage2) grouping all of them?

Thanks!

neusdan, 2018-02-10:
I can't help with your questions. But as your questions are general grub2 questions and not trustedgrub2 specific you should try to get help at the original grub2 project. They probably have a development mailing list for example.

jorgefm, 2018-02-13:
Hi! Thanks for your answer.

Finally, I managed to do what I want using 'grub-mkimage' to generate the core.img file composed by several modules and embedding a grub.cfg file (with the --config parameter). Then 'grub-bios-setup' to setup the device with the grub generated in the first step.

Thanks!

closed 2021-09-28 by securitykernel